nakujaproject/Flight-Control #1: Select microcontroller

SammyOina, 2021-05-28:
Keys:
- Speed
- Size
- Interfaces

Deliverable:
- Presentation
- Wiki page

Pancakem, 2021-06-02:
Hello @SammyOina. Are there guys on this issue?

SammyOina, 2021-06-05:
> Hello @SammyOina. Are there guys on this issue?

Yes there are but i can  add you as well

TheParadox20, 2021-06-12:
The choice of a development board to be used for flight control was based on the following features:

- Number of CPU cores
- CPU architecture
- MC speed
- RAM and Flash memory
- Number of GPIO pins
- Method of programming
- Cost and availability

Factoring the above, the following boards were shortlisted:

1. STM32F103 (Blue pill)
2. ESP32
3. ESP8266
4. Arduino Nano
### Blue pill
The STM32F103C8T6 is a development board developed by STMicroelectronics with an Arm Cortex-M3 as the CPU.
[image]

### ESP32
This system on a chip (SoC) microcontroller developed by espressif and uses the Xtensa LX6 CPU. It has the added capability of Wireless connections and has a hall sensor integrated.
[image]

### ESP8266
The esp8266 is the predecessor of the ESP32.
[image]

### Arduino Nano
This is another board that belongs to the Arduino family. It has an Atmega328P CPU that powers the board
[image]
### The ATmega328P
An alternative to using development boards, a single microcontroller unit may be used. 
[image]
The Atmega328P is the MCU used in Arduino Uno and Arduino Nano. It may be programmed using an Arduino board then used to control a circuit
### Programminga
All the above boards can be programmed with the Arduino IDE or platformIO and C++ language using Arduino syntax
### Specifications
[image]

Mike-Kimani, 2021-06-27:
Thank you @TheParadox20. From your research, ESP32 is our best option.

closed 2021-06-27 by Mike-Kimani