nakujaproject/Flight-Control #7: Research on Extended Kalman Filter

SammyOina, 2021-05-29:
as well as other advanced filters
Deliverable:
- wiki page
- presentation

IanKimoi, 2021-06-03:
> Assigned #7 <https://github.com/nakujaproject/Flight-Control/issues/7> to
> @IanKimoi <https://github.com/IanKimoi>.

Noted.

WashingtonKK, 2021-06-03:
Noted

WashingtonKK, 2021-07-08:
We recommend that we use the kalman filter as implemented in N-1, this is due to the promising results given by the kalman filter in N-1 and because we had not completely utilized the kalman filter in N-1.

WashingtonKK, 2021-07-08:
We use the Kalman filter and not the extended Kalman filter.

> We recommend that we use the kalman filter as implemented in N-1, this is due to the promising results given by the kalman filter in N-1 and because we had not completely utilized the kalman filter in N-1.