nakujaproject/Flight-Control #13: Review Flight control Computer

SammyOina, 2021-11-27:
Review:
- [ ] Schematic
- [ ] PCB
Use Datasheets and other material to review connections prior to PCB order placement
Files: https://github.com/nakujaproject/flight-control-computer
Software: EasyEDA

WashingtonKK, 2021-12-20:
👍
Seems good. 
Let the rest do the checking too

0x6f736f646f, 2021-12-20:
- @SammyOina provide I2C connections to the telemetry computer
- Power on LED
- For the decoupling capacitors on the VR shouldn't they be two 100uF and 10nF?
- Check the type of switches you have used, how they are positioned on the PCB. Will it be difficult to operate them?
- For the switch debouncing circuit your bounce time is 66.329ms. You can use a 0.1uF cap to achieve up to 1ms bounce time.
- The USB connector place it on the periphery e.g as shown below
 [image]

SammyOina, 2021-12-26:
> * @SammyOina provide I2C connections to the telemetry computer
> * Power on LED
> * For the decoupling capacitors on the VR shouldn't they be two 100uF and 10nF?
> * Check the type of switches you have [...]

@0x6f736f646f for telemetry we were providing communication via UART, is I2c Preferred for telemetry?

0x6f736f646f, 2021-12-28:
it is faster

SammyOina, 2021-12-29:
> * @SammyOina provide I2C connections to the telemetry computer
> * Power on LED
> * For the decoupling capacitors on the VR shouldn't they be two 100uF and 10nF?
> * Check the type of switches you have [...]

I have changed the capacitors on the VR and the switches

SammyOina, 2021-12-29:
> it is faster

Ok, we've switched from serial to I2C

SammyOina, 2021-12-29:
> * @SammyOina provide I2C connections to the telemetry computer
> * Power on LED
> * For the decoupling capacitors on the VR shouldn't they be two 100uF and 10nF?
> * Check the type of switches you have [...]

Power On Led added

SammyOina, 2021-12-29:
Switches moved
[image]
Usb moved further
[image]

PMunyao, 2022-01-02:
still a noob at easyeda, but from what I've been able to make sense of, it seems good

Mike-Kimani, 2022-01-02:
Since you still have a lot of space I'd advise you to expose rx&tx pins for redundacy purposes.

SammyOina, 2022-01-04:
> Since you still have a lot of space I'd advise you to expose rx&tx pins for redundacy purposes.

Will do

closed 2022-02-02 by SammyOina